ros-visualization/rqt_reconfigure #3: rqt_reconfigure: the layout is pretty broken

dirk-thomas, 2017-04-24:
_From @dirk-thomas on August 16, 2013 22:10_

- [x] left column takes all the space, right column has only minimum width, should be the other way around
- [x] the configuration block per node and vertically spread, should have spacing at the bottom
- [x] each edit row is horizontally spread, should make use of the full width instead
- [ ] the row label is overlapping with the controls when the window is small

_Copied from original issue: ros-visualization/rqt_common_plugins#150_

dirk-thomas, 2017-04-24:
First three bullets fixed with pr #151.

Kukanani, 2017-04-26:
Will these fixes resolve the fixed-width space allocated for the tree in the left panel? The cutoff width does not increase if the left panel is made wider. It results in nested names becoming invisible, as in this picture:
[image]

sloretz, 2025-06-02:
ROS 1 is end-of-life (EOL) as May 31st 2025, so I'm closing this.

[If you still rely on ROS 1, read this page to learn about your options.](https://bit.ly/NoeticEOL)