ros-visualization/rqt_reconfigure #4: rqt_reconfigure not displaying any param

dirk-thomas, 2017-04-24:
_From @fw-timku on February 23, 2017 4:2_

Ubuntu 16.04
Realsense R200 camera

Hi all, I am trying to change the realsense r200 camera settings (lr_gain, lr_exposure etc.) in the r200_params file located in the realsense_camera/cfg folder. I did launch the camera driver through the rgbd nodelet and after which I type the following command 'rosrun rqt_reconfigure rqt_reconfigure' to launch the rqt reconfig window.  However, there is nothing shown in the window. Anyone knows what is the problem?

[image]

_Copied from original issue: ros-visualization/rqt_common_plugins#435_

Kukanani, 2017-04-26:
Hi,

I have seen this issue before. Have you tried dragging the left side panel open?

[image]

csperbeck6, 2018-01-31:
When I execute "rosrun rqt_reconfigure rqt_reconfigure", mine displays the gui window, but displays no parameter values. I do see the expand and collapse buttons. Any ideas?

130s, 2018-01-31:
You might want to go through [troubleshooting](http://wiki.ros.org/rqt_reconfigure#Troubleshoot) section first for this kind of issue on rqt_reconfigure.

csperbeck6, 2018-01-31:
Thanks! I executed "rosrun dynamic reconfigure dynparam list" and it returned nothing so that is probably the issue. I am new to ROS and am not sure how to populate that dynamic list.

jcui65, 2020-02-12:
I also executed "rosrun dynamic reconfigure dynparam list" and it returned nothing. Then what to do next to fix this problem?

yongan007, 2020-02-24:
u should run _rosrun rqt_gui rqt_gui --clear-config_

chaolmu, 2020-03-16:
Hallo, I have the same issue when putting the setup into a if segment. "dynparam list" returns nothing.
However if I put the setup outside the if segment (the only difference is shown in the pics below), the dynamic reconfigure server works fine.  It is proved that the program went inside the if segment, even the callback function has called once. 
[image]
[image]

PS: I would like to setup dynamic reconfigure depending on controller types, so a "if" is necessary.

SuYouge, 2020-04-07:
> I have the same issue my rqt is just empty. I solve it by _rosrun rqt_gui rqt_gui --clear-config_
> 
> and then go _plugin -> configuration -> dynamics configuration_

I met the same issue as @Kukanani , i can`t use `rosrun rqt_reconfigure rqt_reconfigure`  

but `rqt` with `dynamics configuration` selected works fine which really confuses me.

Is there any solution for this issue? 
[image]

130s, 2020-05-13:
I don't have a fix but just another data point as a coworker just ran into this issue and was able to work it around by drag'n'drop suggested in https://github.com/ros-visualization/rqt_reconfigure/issues/4#issuecomment-297530525

I do see `rosrun dynamic reconfigure dynparam list` was NOT empty, in fact I did see the list I expect to see. 

```
$ apt-cache policy ros-kinetic-rqt-reconfigure
ros-kinetic-rqt-reconfigure:
  Installed: 0.5.1-1xenial-20200229-053042+0000
```

sharathsrini, 2020-11-24:
> Hi,
> 
> I have seen this issue before. Have you tried dragging the left side panel open?
> 
> [image]

This works. Thank you so much

ehvenimunis, 2021-05-18:
> Hi,
> 
> I have seen this issue before. Have you tried dragging the left side panel open?
> 
> [image]
This works. Thank you so much