ros-visualization/rqt_reconfigure #5: rqt_reconfigure resize disabled with qt5

dirk-thomas, 2017-04-24:
_From @nxdefiant on April 24, 2017 10:41_

In kinetic the rqt_reconfigure GUI does not allow resize between the left and the right column. This is problematic for e.g. reconfigure params with middle to longer text and reduces the overall usability.

_Copied from original issue: ros-visualization/rqt_common_plugins#450_

nxdefiant, 2017-04-25:
ros-visualization/rqt_common_plugins#417 has additional details.

paulsammut, 2017-09-13:
[image]
This is an extremely annoying problem

dirk-thomas, 2017-09-13:
Please consider to provide a PR for this. The previously proposed patch (ros-visualization/rqt_common_plugins#417) didn't work.

paulsammut, 2017-09-14:
Anything on this? I tried inserting this in node_selector_widget.py:

`self._node_selector_view.header().setSectionResizeMode(0, QHeaderView.ResizeToContents)  `

but i get a segfault on launch. Xenial with Kinetic. Is there any quick line I can inject somewhere that will let me see the text in the tree?

cottsay, 2017-09-14:
Ack, I'll bump this to the top of my queue.

nxdefiant, 2017-09-14:
paulsammut, did you have any dynreconf enabled nodes running when it was crashing?

paulsammut, 2017-09-14:
@nxdefiant, I did not! 

Lo and behold, I ran it after loading up some dynamic reconfigure nodes and it works! 

[image]

nxdefiant, 2017-09-14:
As I thought, protecting the line with
`if self._node_selector_view.header().length() > 0:`
should do it.

paulsammut, 2017-09-14:
Just FYI and in an effort to make a better rqt_reconfigure. With that fix, if you run rqt_reconfigure before launching any dynreconf nodes, the tree will not be properly resized. But, I am very very happy with this

julianraheema, 2018-03-15:
For the binary rqt_reconfigure. go to the /opt/ros/kinetic/share/rqt_reconfigure/resource directory and change node_selector.ui line 95 from false to true.

cottsay, 2018-04-19:
I believe this was addressed by #11, and should be in the current release. Please reopen if this is still a problem.

closed 2018-04-19 by cottsay