ros-visualization/rqt_reconfigure #12: [enhancement] rqt_reconfigure : setting decimal value to double parameters

ttdm, 2017-08-24:
Hello, 

I believe that there is a non desired behavior when trying to set decimal values for doubles. 

Let's say that i have a double which range is [0,1], i believe we should be able to set it's value to anything in [0,1] in a few different ways: 
1/ using the cursor with a interval_length/100 step
2/ directly writing the number ex : 0.123
3/ copying another value in the gui then pasting it there
4/ writing the number in scientific manner ex : 123e-3

On those 4 ways to set the value, 2/ is what's seem the most commonly used but does not work at the moment. I cannot directly write for example "0.123" which is quite unexpected I must say and which is probably a non desired behavior. 

Looking a little bit at different responses of the gui, I think what doesn't work is to add any "." in the value. For example this makes that it is impossible to copy/paste a value from another double parameter ( /3 ), 

Since nearly all symbol are not allowed in the box, my guess is that you are checking the value in the box with a regEx and that there is a little mistake in the regex. I believe there is a mistake because "," is accepted in the box but when validated, the gui crash with the following error : 
me@laptop-293:~$ rosrun rqt_reconfigure rqt_reconfigure 
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/param_editors.py", line 354, in _text_changed
    self._update_paramserver(float(self._paramval_lineEdit.text()))
ValueError: invalid literal for float(): 0,1
So to me the issue is that "," is accepted instead of "." in the box. 

I tried to use /4 and it actually works so this issue isn't really important ( but it remains that it's probably an undesired behavior ).
 
I must add that i'm a french user and that most of the time my keybaord is in "azerty" configuration. But i tried to put it into qwerty and i still had the same issue.  I also tried it from 2 different and that I had the exact same behavior. 

I'm curious to know if I'm the only one experimenting this issue. 
And thanks a lot to the one who will have the skill and the time to correct it !

cottsay, 2017-08-25:
Thanks for reporting this, @ttdm. I'll try to take a look this weekend.

cottsay, 2018-04-19:
Hi @ttdm - I believe #21 should fix this problem. Expect it in the next release.

closed 2018-04-19 by cottsay

ttdm, 2018-04-19:
Hi, thanks for the update. I just tried to switch to English locale and it works so it should indeed be the same issue ;)   

Thanks for the fix.