ros-visualization/rqt_reconfigure #18: rqt_reconfigure does not display any parameters

csperbeck6, 2018-01-31:
When I execute "rosrun rqt_reconfigure rqt_reconfigure", the gui window pops up, but I see no parameters to change. Am I missing some installation or configuration to do this? This is what appears when I execute the command:
[image]

130s, 2018-01-31:
How is this different from https://github.com/ros-visualization/rqt_reconfigure/issues/4?

cottsay, 2018-04-19:
@csperbeck6, are you still having this issue? The [troubleshooting guide](http://wiki.ros.org/rqt_reconfigure#Troubleshoot) talks about using the command line dynamic reconfigure client to confirm what `rqt_reconfigure` is showing. Can you give that a shot?

cottsay, 2020-04-24:
After getting no response from @csperbeck6 for quite some time, I'm going to close this issue. If you're still having problems and have time to work with me to reproduce them, please feel free to re-open this issue.

closed 2020-04-24 by cottsay