ros-visualization/rqt_reconfigure #19: Cannot select any node if many nodes are running

furushchev, 2018-02-25:
Hi, I'm a user of PR2 and have a problem to set params with `rqt_reconfigure`.
When clicking any node in the tree view, the GUI often (really often!) hangs and goes black, and the only thing I can do is to reboot `rqt_reconfigure` node.

I can also see a lot of errors when this node is started like as follows:

```
Exception in thread Thread-114:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/home/furushchev/ros/kinetic/src/ros-visualization/rqt_reconfigure/src/rqt_reconfigure/treenode_qstditem.py", line 67, in run
    self._param_name_raw))
ROSException: timeout exceeded while waiting for service /kinect_head_external/rgb_rectify_mono/set_parametersTreenodeQstdItem. Couldn't connect to /kinect_head_external/rgb_rectify_mono
```

[image]

furushchev, 2018-04-11:
@cottsay kindly ping to the maintainer. Any proceedings?

cottsay, 2018-04-11:
A apologize, @furushchev. A couple of other items are pending for this package now, including a release for Melodic. I'll prioritize this fix and another, followed by a Melodic, Lunar and Kinetic.

closed 2018-04-19 by cottsay

furushchev, 2018-04-19:
@cottsay No problem! Thank you for your hard work! 👍

PR #20 (2018-04-19): lazy creating of dynamic_reconfigure client for each node
This fixes #19.

In current implementation, the `rqt_reconfigure` node tries to connect all nodes available when it starts.
If I understand ROS ecosystem correctly, each connection uses one socket, so if there are many nodes running in one ROS master such as PR2 robot, it can use up all available sockets.

In this pull request, I fixed codes not to connect to all nodes on the start of the node and only connect when any node is selected instead.

I'm not sure if there is any side effect on this change, but at least it works on my laptop, [... cut]