ros-visualization/rqt_reconfigure #22: Release into Melodic

clalancette, 2018-04-10:
It looks like all of the dependencies for this package are available in Melodic, so it would be great to get it released there.  Thanks in advance.

130s, 2018-04-19:
@cottsay I can offer to co-maintain if that helps (as I actively use this package), and help the release. If that's fine I just need a write access to this and release repo.

clalancette, 2018-04-19:
Ping?  Can we get this released into Melodic, please?  Thanks.

cottsay, 2018-04-19:
I just got the other fixes merged yesterday. The melodic release is complete: https://github.com/ros/rosdistro/pull/17545

@130s, you actually used to be a maintainer for this package. If you'd like to be again, go ahead and open a PR to add yourself to the package.xml, and I'll grant you access to the repos.

closed 2018-04-19 by cottsay