ros-visualization/rqt_reconfigure #26: Qt platform plugin "xcb"

pedrombmachado, 2018-08-04:
Dear team,
I am getting the following error:
```
$ rosrun rqt_gui rqt_gui
This application failed to start because it could not find or load the Qt platform plugin "xcb".

Reinstalling the application may fix this problem.
```
Details:
Ubuntu 18.04 LTS x64
ROS Melodic

cottsay, 2018-09-12:
A couple of things:
1. I can't reproduce this issue
2. This is the `rqt_reconfigure` package, not the `rqt_gui` package

Here is the dependency chain:
Start with `ros-melodic-rqt-gui`
...which requires `ros-melodic-qt-gui`
...which requires `ros-melodic-python-qt-binding`
...which requires `python-pyqt5`
...which requires `libqt5gui5`
...which provides `libqxcb.so`

So unless there is a bug in packaging somewhere, this issue is probably specific to your environment. Can you run any other Qt applications?

cottsay, 2020-04-23:
Having no response from @pedrombmachado for well over a year, I'm going to close this issue. Please feel free to re-open it if you have time to help me reproduce your issue.

closed 2020-04-23 by cottsay