ros-visualization/rqt_reconfigure #28: Mouse wheel button unintuitively moves slide bar

130s, 2018-10-16:
# Current behavior

As per title. The slide bar that has a focus moves, when a mouse's wheel rolls.

# Expected behavior

I think it's more intuitive if the vertical scroll bar moves.

(I've seen my coworkers keep complaining, without filing a ticket.)

# Version

```
$ apt-cache policy ros-indigo-rqt-reconfigure
ros-indigo-rqt-reconfigure:
  Installed: 0.4.10-0trusty-20181006-073800-0800
  Candidate: 0.4.10-0trusty-20181006-073800-0800
  Version table:
 *** 0.4.10-0trusty-20181006-073800-0800 0
        500 http://packages.ros.org/ros-shadow-fixed/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status
```

130s, 2018-10-16:
I'm interested in making a patch to realize the expected behavior above, unless someone has other ideas.

cottsay, 2018-10-16:
Sounds like an OK change to make. I don't recall when that behavior was introduced, but I can see how it might lead to undesired (and rapid) changes.

cottsay, 2020-04-23:
It might be possible to make the sliders respond to horizontal scrolling (for devices which can do that) instead of vertical scrolling. I would find that to be more intuitive.

PR #73 (2020-05-08): Don't process scroll events unless specifically focused
It can be really easy to accidentally change these values while scrolling along in the editor. This change makes it necessary to focus (clicking on, tabbing to, etc) a widget before the scroll wheel can change it.

Closes #28
closed 2020-05-08 by cottsay