ros-visualization/rqt_reconfigure #52: [ROS2]rqt_reconfigre UI hangs

xhuan28, 2019-08-12:
ROS2 rqt_reconfigure is not stable. Either UI hangs or can not launched successfully.
Some error logs as follows:
[ERROR] [rqt_reconfigure]: Reconfigure GUI cannot connect to master.
Traceback (most recent call last):
  File "/home/intel/workspace/ros2_rqt_ws/build/rqt_gui_py/src/rqt_gui_py/rclpy_spinner.py", line 32, in run
Traceback (most recent call last):
    executor.spin_once(timeout_sec=1.0)
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/node_selector_widget.py", line 444, in _refresh_nodes
  File "/home/intel/workspace/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/executors.py", line 695, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/home/intel/workspace/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/executors.py", line 649, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/home/intel/workspace/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/executors.py", line 532, in _wait_for_ready_callbacks
    self._prune_nodetree_pernode()
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/node_selector_widget.py", line 431, in _prune_nodetree_pernode
    self._context.handle)
RuntimeError: Failed to initialize wait set: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /home/intel/workspace/ros2_ws/src/ros2/rcl/rcl/src/rcl/wait.c:130
    raise e  # TODO Make sure 'raise' here returns or finalizes func.
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/node_selector_widget.py", line 428, in _prune_nodetree_pernode
    nodes = find_nodes_with_params(self._context.node)
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/param_api.py", line 133, in find_nodes_with_params
    for node_name, ns in node.get_node_names_and_namespaces()
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/param_api.py", line 130, in <lambda>
    filter(lambda node_name: _has_params(node, node_name),
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/param_api.py", line 122, in _has_params
    client.wait_for_service()
  File "/home/intel/workspace/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/client.py", line 164, in wait_for_service
    return self.service_is_ready()
  File "/home/intel/workspace/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/client.py", line 144, in service_is_ready
    return _rclpy.rclpy_service_server_is_available(capsule)
RuntimeError: Failed to check service availability: rcl node's context is invalid, at /home/intel/workspace/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:476

The above exception was the direct cause of the following exception:

SystemError: <class 'PyQt5.QtCore.QChildEvent'> returned a result with an error set

The above exception was the direct cause of the following exception:

SystemError: <class 'PyQt5.QtCore.QChildEvent'> returned a result with an error set

The above exception was the direct cause of the following exception:

SystemError: <class 'PyQt5.QtCore.QChildEvent'> returned a result with an error set

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/intel/workspace/ros2_rqt_ws/install/qt_gui/lib/python3.6/site-packages/qt_gui/plugin_handler.py", line 425, in _check_close
    if len(self._widgets) + len(self._toolbars) == 0:
SystemError: <built-in function len> returned a result with an error set

The above exception was the direct cause of the following exception:

SystemError: <class 'PyQt5.QtCore.QEvent'> returned a result with an error set

The above exception was the direct cause of the following exception:

SystemError: <class 'PyQt5.QtWidgets.QWidget'> returned a result with an error set

The above exception was the direct cause of the following exception:

SystemError: <class 'PyQt5.QtGui.QCloseEvent'> returned a result with an error set

The above exception was the direct cause of the following exception:

SystemError: PyEval_EvalFrameEx returned a result with an error set

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/intel/workspace/ros2_other_ws/install/rqt_reconfigure/lib/rqt_reconfigure/rqt_reconfigure", line 10, in <module>
    sys.exit(main.main(sys.argv, standalone=plugin, plugin_argument_provider=ParamPlugin.add_arguments))
  File "/home/intel/workspace/ros2_rqt_ws/build/rqt_gui/src/rqt_gui/main.py", line 59, in main
    plugin_argument_provider=plugin_argument_provider)
  File "/home/intel/workspace/ros2_rqt_ws/install/qt_gui/lib/python3.6/site-packages/qt_gui/main.py", line 646, in main
    return app.exec_()
SystemError: <built-in method exec_ of QApplication object at 0x7f1f7ca7f168> returned a result with an error set

cottsay, 2019-08-12:
The backtrace references several packages, and the error seems to originate outside of `rqt_reconfigure`. Please post the SHA or version number of these packages:
- rcl
- rclpy
- rqt_gui_py

xhuan28, 2019-08-13:
- rcl: commit 7984f944236a40f05e23c4fd758bca6279939c49

- rclpy: commit 113a8f27e236022d4126899829e51c431830872c

- rqt_gui_py:  commit e05bb51244c86888134c2ad3a661fed91246bf46

dirk-thomas, 2020-01-14:
Without steps to reproduce the problem there is not much we can do about it. I will close the ticket for now. Please feel free to comment with the necessary information and the ticket can be reopened.

closed 2020-01-14 by dirk-thomas