ros-visualization/rqt_reconfigure #53: [ROS2]Cannot get or set parameters with the type of PARAMETER_INTEGER_ARRAY

xhuan28, 2019-08-14:
I have several parameters whose type are PARAMETER_INTEGER_ARRAY. It seems that rqt_reconfigure hasn't support this type yet, since I was trying to launch rqt_reconfigure, it failed with the following error messages:

> Traceback (most recent call last):
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/node_selector_widget.py", line 277, in _selection_changed_slot
    self._selection_selected(index_current, rosnode_name_selected)
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/node_selector_widget.py", line 228, in _selection_selected
    raise e
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/node_selector_widget.py", line 226, in _selection_selected
    item_widget = item_child.get_param_client_widget()
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/treenode_qstditem.py", line 94, in get_param_client_widget
    self._context, self._raw_param_name
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/param_client_widget.py", line 116, in __init__
    self._param_client.get_parameters(param_names),
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/param_api.py", line 89, in get_parameters
    for name, value in zip(names, get_params_response.values)
  File "/home/intel/workspace/ros2_other_ws/build/rqt_reconfigure/src/rqt_reconfigure/param_api.py", line 89, in <listcomp>
    for name, value in zip(names, get_params_response.values)
  File "/home/intel/workspace/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/parameter.py", line 121, in from_parameter_msg
    return cls(param_msg.name, type_, value)
  File "/home/intel/workspace/ros2_ws/install/rclpy/lib/python3.6/site-packages/rclpy/parameter.py", line 132, in __init__
    raise ValueError("Type '{}' and value '{}' do not agree".format(type_, value))
ValueError: Type 'Type.INTEGER_ARRAY' and value 'array('q', [640, 480])' do not agree
[INFO] [PluginContainerWidget]: Connection to node=/realsense failed:
Type 'Type.INTEGER_ARRAY' and value 'array('q', [640, 480])' do not agree
[ERROR] [rqt_reconfigure]: Connection to node=/realsense failed:
Type 'Type.INTEGER_ARRAY' and value 'array('q', [640, 480])' do not agree

On the other hand, I was trying to get/set parameters by ros2 param. Everything works well.

>$ros2 param set my_node my_param [640,480]
Set parameter successful
>$ros2 param get my_node my_param
Integer values are: array('q', [640, 480])

gonzodepedro, 2019-09-14:
Created #56 and ros2/rclpy#424

dirk-thomas, 2019-10-10:
What is the status if this?

gonzodepedro, 2019-10-11:
It is in my backlog to work on. I created issues #56 and ros2/rclpy#424 which together will solve this issue. I still need to created PR to fix those. I'm expecting to make progress on those next week.

valioiv, 2019-10-28:
@gonzodepedro ,
the array issue in the constructor seems fixed and merged now:
https://github.com/ros2/rclpy/pull/444

christianrauch, 2022-12-11:
This should be solved by https://github.com/ros-visualization/rqt_reconfigure/pull/108. Please check and close.

PR #108 (2022-10-04): Add support for array types
Closes #53 and #56
closed 2022-12-12 by xhuan28