ros-visualization/rqt_reconfigure #56: Add support for array types

gonzodepedro, 2019-09-14:
UI Add EditorWidgets in param_editors.py to support the following types:

- byte_array_value
- bool_array_value
- integer_array_value
- double_array_value
- string_array_value

christianrauch, 2022-12-11:
This should be solved by https://github.com/ros-visualization/rqt_reconfigure/pull/108. Please check and close.

gordon-n-stevenson, 2024-05-21:
Hi @christianrauch, i'm running rqt_reconfigure in humble and #108 seems to work having patched param_editor over to my working build. 

Is it possible to push this commit to not only rolling but to humble (and others I guess) as well?

christianrauch, 2024-05-21:
I guess you have to ask in https://github.com/ros-visualization/rqt_reconfigure/pull/108 if this can be backported to humble, or better, just send a PR that backports that fix.

gordon-n-stevenson, 2024-05-27:
Thanks! Have raised a PR to do this https://github.com/ros-visualization/rqt_reconfigure/pull/145