ros-visualization/rqt_reconfigure #58: [ROS2] Tool does not list any nodes and params

valioiv, 2019-10-21:
After build and run the latest Dashing branch the GUI does not list any information at all. Is it a known issue?
What is the pending work to be done in order to have it working. I'm open to contribute if you can give some directions how I could help...

valioiv, 2019-10-21:
@asenisa12, correct, this exactly what I observe - even if I already have running node with declared parameters...

gonzodepedro, 2019-10-22:
I'm not able to reproduce it. Please, provide more information. 

- How did you run it ? 
- What information was expected?
- Provide the output of 'ros2 param list'
- rmw implementation used

valioiv, 2019-10-23:
The problem was we were trying to run it with pySide but not pyQT. With pyQT rqt_reconfigure works fine. I'm closing the issue. Thank you!

closed 2019-10-23 by valioiv