ros-visualization/rqt_reconfigure #59: New release containing python3 fixes

timonegk, 2019-10-22:
Since release 0.4.10, some changes regarding python3 compatibility have been made. It would be nice to have a python3 compatible release.

dirk-thomas, 2019-10-22:
> some changes regarding python3 compatibility have been made

Can you please point to the changes you are referring to.

cottsay, 2019-10-22:
I think @timonegk is referring to all of the changes that we made to the ROS 1 package during the course of development for ROS 2: https://github.com/ros-visualization/rqt_reconfigure/compare/0.4.10..master

The changes were brought back, smoke tested and passed CI builds, but they haven't been released.

It seems appropriate to cut a 0.5.0 release for Noetic, but I'm not sure we should bump Melodic yet. Even thought Melodic is Python 2, there are fixes that came along with the Python 3 compatibility, so there would be value in a Melodic release, but I'd like to see more testing before we do that.

dirk-thomas, 2019-10-22:
I agree that we should release a new version but I still would like to know which specific change is relevant for Python 3 compatibility.

cottsay, 2019-10-22:
> I still would like to know which specific change is relevant for Python 3 compatibility.

Probably #30

timonegk, 2019-10-22:
I am indeed referring to #30 and the issue mentioned in #28 that was (accidentally?) closed in #35.

timonegk, 2019-10-25:
Thanks for the release!

The changes made in #30 were unfortunately removed in #35:
https://github.com/ros-visualization/rqt_reconfigure/commit/e55cfa1c9b2ba1ec97626bc330d6e97f3237f795#diff-3a7c204e9c16a9027d10eaab54a32b69L72

cottsay, 2019-10-25:
Oh no! I'll fix that today. Thanks for following up.

timonegk, 2019-10-25:
Thanks a lot!

closed 2019-10-25 by timonegk