ros-visualization/rqt_reconfigure #64: rqt hangs when node previoulsy shown is stopped and clicked

dirk-thomas, 2020-01-14:
See https://answers.ros.org/question/339142/rqt-freezes-when-opening-dynamic-reconfigure/

QQting, 2020-06-24:
I also encountered this issue. Are there any plans to fix it?

doronhi, 2020-07-07:
Also encountered that issue while running the turtlesim_node example on ubuntu18.04 with eloquent distro.

kdesorme, 2020-09-02:
I also encoutered this issue and can't find a solution. Ubuntu 18.04, ros melodic.

mjeronimo, 2020-10-16:
Once the node is stopped/goes away, the list box still has the item in it. Then, when the user clicks that item, the code sets up a parameter client to the node (providing the parameter services) and invokes the parameter client to list the parameters. It is using a synchronous call to do this (with no timeout) and hangs because the node and its parameter services are no longer there.

Basically, the current design relies upon the user the hit the Refresh button to remove stale nodes and unfortunately hangs if one is used.

A couple ways to address this:

* There could be a thread that automatically refreshes the list, say every second, so that stale nodes are removed.
* Even better, @dirk-thomas is there a way to receive an event when a new node is created or destroyed in the system?

dirk-thomas, 2020-10-16:
> is there a way to receive an event when a new node is created or destroyed in the system?

Since you have cross posted the same question on Slack where you got an answer please summarize it here in order to be public.

mjeronimo, 2020-10-16:
Sure. The suggestion from @wjwwood is to wait for graph events:

> You can wait for graph events. They will not tell you what happened, but it can be a notice that tells you if you need to recheck everything. See: https://github.com/ros2/rclcpp/blob/018cfaa2197b94ed1517cc2860758e1b8db02bdb/rclcpp/src/rclcpp/client.cpp#L127-L182

> You could wait on a graph condition in a loop and reload the list of nodes each time it changes.

> The rmw version has better docs: https://github.com/ros2/rmw/blob/e02978028f1756cc4c057d06f462ceac8acafcab/rmw/include/rmw/rmw.h#L201-L253

I don't believe this functionality (get_graph_event() and wait_for_graph_change()) are available in the rclpy interface yet and would need to be added. If we can add this functionality to rclpy and then monitor for graph changes (which include nodes coming and going) we can update the list of nodes in rqt_reconfigure, avoiding the need for the refresh button and avoiding crashes due to stale node names in the list.

PR #98 (2021-02-02): Use timeouts in service calls to avoid hangs
Don't assume that all ROS nodes support the parameter services. Instead explicitly look for the ListParameters service.
    
Also, don't assume that all nodes are spinning. Instead of using synchronous service calls (which don't allow for a timeout and will hang if the node is not spinning to process its messages) using an async call, which returns a future that can be used with rclpy.spin_until_future_complete, which supports a timeout.

Note: Parameters can be dynamically added to a node. So, nodes that support the parameter services but currently [... cut]
commit 3471cfef (2021-02-02): Use timeouts in service calls to avoid hangs (#98)

* Use async service calls to avoid hangs

Don't assume that all ROS nodes support the parameter services. Instead
explicitly look for the ListParameters service.

Also, don't assume that all nodes are spinning. Instead of using synchronous
service calls (which will hang if the node is not spinning to process its
messages) using an async call with a timeout.

Fixes #97, fixes #96, fixes #64

* Use leading underscore on private function

* Remove the sleep in the constructor and handle auto-refresh in a different PR

* Cancel the future in case [... cut]
closed 2021-02-02 by mjeronimo