ros-visualization/rqt_reconfigure #68: Failed to run the executable

yossioo, 2020-03-03:
Hello

I am trying to launch the `rqt_reconfigure` node and it does not show up.

To examine it I have installed fresh copy of Ubuntu 18.04.4 on virtual machine, and installed `ros-eloquent-desktop` according to the [latest instructions](https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Install-Debians/), everything is updated to this date (March 3, 2020).

Executing `ros2 run rqt_reconfigure rqt_reconfigure` shows no output and no window opens.

Executing `ros2 run rqt_reconfigure rqt_reconfigure  --ros-args --log-level debug` shows some debug info, with repeating lines:
```
[DEBUG] [rcl]: Initializing timer with period: 1000000000ns
[DEBUG] [rcl]: Updated timer period from '1000000000ns' to '1000000000ns'
[DEBUG] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '1' timers, '1' clients, '6' services
[DEBUG] [rcl]: Waiting with timeout: 0s + 999878540ns
[DEBUG] [rcl]: Timeout calculated based on next scheduled timer: true
[DEBUG] [rmw_fastrtps_shared_cpp]: looking for subscriber topic: /_ros2cli_daemon_0/list_parametersRequest, number of matches: 0
[DEBUG] [rmw_fastrtps_shared_cpp]: looking for subscriber topic: /_ros2cli_daemon_0/list_parametersRequest, number of matches: 0
[DEBUG] [rmw_fastrtps_shared_cpp]: looking for subscriber topic: /_ros2cli_daemon_0/list_parametersRequest, number of matches: 0
[DEBUG] [rmw_fastrtps_shared_cpp]: looking for subscriber topic: /_ros2cli_daemon_0/list_parametersRequest, number of matches: 0
[DEBUG] [rcl]: Timer in wait set is ready
[DEBUG] [rcl]: Timer canceled
[DEBUG] [rcl]: Initializing timer with period: 1000000000ns
[DEBUG] [rcl]: Updated timer period from '1000000000ns' to '1000000000ns'
[DEBUG] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '1' timers, '1' clients, '6' services
[DEBUG] [rcl]: Waiting with timeout: 0s + 999847316ns
[DEBUG] [rcl]: Timeout calculated based on next scheduled timer: true
[DEBUG] [rmw_fastrtps_shared_cpp]: looking for subscriber topic: /_ros2cli_daemon_0/list_parametersRequest, number of matches: 0
[DEBUG] [rmw_fastrtps_shared_cpp]: looking for subscriber topic: /_ros2cli_daemon_0/list_parametersRequest, number of matches: 0
[DEBUG] [rmw_fastrtps_shared_cpp]: looking for subscriber topic: /_ros2cli_daemon_0/list_parametersRequest, number of matches: 0
[DEBUG] [rmw_fastrtps_shared_cpp]: looking for subscriber topic: /_ros2cli_daemon_0/list_parametersRequest, number of matches: 0
[DEBUG] [rcl]: Timer in wait set is ready
[DEBUG] [rcl]: Timer canceled
```

This happens both if there nodes running and if there are not.

Any ideas what could cause this behavior?
I am experiencing similar behavior on 18.04.4 with ROS2 Dashing as well...

gonzodepedro, 2020-03-05:
Please, run 'ros2 parameters' and post the output.

yossioo, 2020-03-05:
> Please, run 'ros2 parameters' and post the output.

I am not quite sure what did you mean by this since I am unfamiliar with this command, but here it is:
```
yossi@laptop-yo ~:
$ ros2 parameters
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'parameters' (choose from 'action', 'bag', 'component', 'daemon', 'extension_points', 'extensions', 'launch', 'lifecycle', 'msg', 'multicast', 'node', 'param', 'pkg', 'run', 'security', 'service', 'srv', 'topic')
```

In case you are interested in `ros2 param list` output (running `examples_rclcpp_minimal_publisher publisher_lambda` in another terminal):
```
yossi@laptop-yo ~:
$ ros2 param list
/minimal_publisher:
  use_sim_time
```

gonzodepedro, 2020-03-05:
Does that happen only when you have a "ros2 daemon" process running?

```
ps -fea | grep ros2
```

Shows a process like:

```
/usr/bin/python3 /opt/ros/eloquent/bin/_ros2_daemon --rmw-implementation rmw_fastrtps_cpp --ros-domain-id 0
```

Thanks for your help

yossioo, 2020-03-05:
Please see the [edit](https://github.com/ros-visualization/rqt_reconfigure/issues/68#issuecomment-595357139) above.

Regarding the daemon:
> Does that happen only when you have a "ros2 daemon" process running?
> 
> ```
> ps -fea | grep ros2
> ```
> 
> Shows a process like:
> 
> ```
> /usr/bin/python3 /opt/ros/eloquent/bin/_ros2_daemon --rmw-implementation rmw_fastrtps_cpp --ros-domain-id 0
> ```
> 
> Thanks for your help

I do have the ros2 daemon running:
```
$ ps -fea | grep ros2
yossi    10528     1  0 19:38 pts/3    00:00:00 /usr/bin/python3 /opt/ros/dashing/bin/_ros2_daemon --rmw-implementation rmw_fastrtps_cpp --ros-domain-id 0
yossi    19836  8323  0 20:10 pts/3    00:00:00 grep --color=auto ros2
```

Should I try to stop it or something?

gonzodepedro, 2020-03-05:
Please ignore `ros2 parameters`, see my [edit](https://github.com/ros-visualization/rqt_reconfigure/issues/68#issuecomment-595336153).

> Should I try to stop it or something?

Please stop it, and then run: `ros2 run rqt_reconfigure rqt_reconfigure`

yossioo, 2020-03-05:
That would do the trick:
```
yossi@laptop-yo ~:
$ ros2 daemon stop
The daemon has been stopped
yossi@laptop-yo ~:
$ ros2 run rqt_reconfigure rqt_reconfigure
```
Opens the reconfigure GUI and it works as expected.

Does having the daemon not running affects something crucial? It seems that the reconfigure GUI discovers newly created nodes and everything works fine. 
Topic and node discovery work, ( it seems that it did relaunch the daemon).

Since the daemon relaunched do I have to stop it again next time I launch the rqt_reconfigure?

gonzodepedro, 2020-03-05:
> Does having the daemon not running affects something crucial? 
It is used by the CLI tools, exactly how is not clear to me, but I would say it is required

> Since the daemon relaunched do I have to stop it again next time I launch the rqt_reconfigure?
We just find a bug!  rqt_reconfigure must work along with the daemon. I'll report it a add this issue as a reference.
Stopping the daemon may be a workaround until the bug gets fixed.

Thanks for your help on tackling this.

yossioo, 2020-03-05:
Sure thing, hope to see the fix soon!

mjeronimo, 2021-01-29:
When starting up, rqt_reconfigure, for each node it discovers in the system, in order to determine if the has parameters or not, invokes the list_parameters service for that node. It makes the service call without a timeout. If it discovers and calls a node that is no longer responsive (for example, if the node has crashed or if the node is not spinning and processing messages), then the call will hang. 

I don't see how the ros2 daemon would affect rqt_reconfigure, but an unresponsive node would prevent rqt_reconfigure from proceeding and displaying its UI (see #68).

mjeronimo, 2021-02-03:
@yossioo Could you check if #98 fixes this issue? Just pull the latest code on the Dashing branch and give it a try.

yossioo, 2021-02-03:
> @yossioo Could you check if #98 fixes this issue? Just pull the latest code on the Dashing branch and give it a try.

It seems to be working better, it does not freeze when Refresh pressed and runs without a problem.
Even when the `ros2 daemon` is running :)

closed 2021-02-03 by yossioo