ros-visualization/rqt_reconfigure #71: Release rqt_reconfigure to ROS Noetic?

sloretz, 2020-04-23:
Would you mind releasing `rqt_reconfigure` to ROS Noetic? It looks like all of its dependencies have been released, and recursively 21 repos need it.

I'm not sure how much work it will need. The [Noetic Migration Guide](http://wiki.ros.org/noetic/Migration) or the [guide about transitioning ROS packages to Python 3](http://wiki.ros.org/UsingPython3/) may be useful.

cottsay, 2020-04-23:
I'll make this happen today.

Related to #70

sloretz, 2020-05-09:
@cottsay friendly ping; mind making a Noetic release?

cottsay, 2020-05-09:
Released 0.5.2: https://github.com/ros/rosdistro/pull/24798

Note that the parameter save button has regressed Python 3 until ros/dynamic_reconfigure#154 is resolved.

closed 2020-05-09 by cottsay