ros-visualization/rqt_reconfigure #92: rqt_reconfigure crashes with IndexError

mallela, 2020-07-22:
I encounter the following error logs when I create a unique pointer to a dynamic_reconfigure::Server and a nodehandle as a param. 

```
PluginManager._load_plugin() could not load plugin "rqt_reconfigure/Param":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 101, in load
    return class_ref(plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/param_plugin.py", line 51, in __init__
    self._plugin_widget = ParamWidget(context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/param_widget.py", line 100, in __init__
    self._nodesel_widget = NodeSelectorWidget(self, rp, self.sig_sysmsg)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/node_selector_widget.py", line 97, in __init__
    self._update_nodetree_pernode()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/node_selector_widget.py", line 332, in _update_nodetree_pernode
    node_name_grn, TreenodeQstdItem.NODE_FULLPATH)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/treenode_qstditem.py", line 87, in __init__
    self._set_param_name(grn_current_treenode)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_reconfigure/treenode_qstditem.py", line 235, in _set_param_name
    self._toplevel_treenode_name = self._list_treenode_names[0]
IndexError: list index out of range
```

cottsay, 2020-07-22:
Hi @mallela -

From what I can tell of your backtrace, you're using a version of `rqt_reconfigure` prior to 0.5.0. The latest version of `rqt_reconfigure` in ROS Kinetic is 0.5.3. It is always a good idea to make sure you're using an up-to-date version of software before reporting bugs. Please try updating to see if your issue is resolved.

If your issue is not resolved, please post the results of this command after the `rqt` failure:
```
rosnode list
```

rolaechea, 2020-10-02:
I have created a patch that fixes the issue. This issue happens even with the latest rqt_reconfigure when a ros node has not shut down properly or has segfaulted.

rolaechea, 2020-10-19:
Is there anything else I can do to help you @cottsay  @mjeronimo  be able to review the pull request ? cc @mallela

mjeronimo, 2020-10-19:
@rolaechea Sorry for the delay. I've been investigating the source of problems like these, which all come down to having rqt_reconfigure's list of nodes out of sync with the nodes actually running. Unfortunately, rqt_reconfigure uses the Refresh button to manually update its list. Instead, it should use an automated method to monitor for active nodes (see 
https://github.com/ros-visualization/rqt_reconfigure/issues/64 for a bit of discussion on this). Meanwhile, util we can fix this correctly, we may need workarounds like this one. I'm OK with this one for now, but would like to fix the core issue.

rolaechea, 2020-11-16:
Just a heads up @mjeronimo  that the function find_reconfigure_services returns as one of the "reconfigurable" services/nodes, the empty string "" whenever there is a service called "/set_parameters". This was the root cause of this crash. When the crash occurred I listed the results of roservice list and saw that there is one called "/set_parameters". Because of the way find_reconfigure_services is written, this results in "" being included as list of reconfigurable services/nodes:

avidbots@1.5.1-rc0:~/Dev/robot-dev/repos/autonomy/ros_indigo/src$ rosservice list | grep -E /set_parameters$ 
/amcl/set_parameters
...
**/set_parameters**

``
def find_reconfigure_services():
    import rosservice
    return sorted([s[:-len('/set_parameters')] for s in rosservice.get_service_list() if s.endswith('/set_parameters')])

mjeronimo, 2020-11-18:
Thanks for the background info, @rolaechea. Very helpful.

Leeable, 2021-04-02:
solution in ROS melodic :
run the command    **sudo gedit /opt/melodic/lib/python2.7/dist-package/rqt_reconfigure/node_selector_widegt.py**

and find the line of 325, replace original one :    if node_name_grn in self._nodeitems or node_name_grn =='':

that'all  ~ enjoy rqt_reconfigure