ros-visualization/rqt_reconfigure #110: [ROS2] Listen for parameter events

bjsowa, 2022-03-19:
When a parameter is changed externally (for example by using `ros2 param set`), the `rqt_reconfigure` does not update the values, unless the node is reselected. Similarly, the plugin does not react when a node declare a new parameter or undeclares one.

This can be easily improved by subscribing to the global `/parameter_events` topic, filtering the messages for only the selected nodes and updating the UI according to the events.

berndpfrommer, 2022-03-21:
I second that feature request. In particular refreshing the values would already help my use case, where a device driver may not be able to change the value of a parameter to the exact number set by the rqt_reconfigure gui. Without rqt_reconfigure updating on parameter change, the device driver has no way of providing feedback to the rqt_reconfigure gui about what the actual current parameter setting is.

bjsowa, 2023-05-15:
I incorrectly assumed that `rqt_reconfigure` does not subscribe on the `/parameter_events` topic while in fact it does and it handles the events here:

https://github.com/ros-visualization/rqt_reconfigure/blob/rolling/src/rqt_reconfigure/param_client_widget.py#L143-L159

The reason it does not work is related to #126. With this PR, the parameter values are correctly being updated.

closed 2023-11-08 by bjsowa