ros-visualization/rqt_reconfigure #130: Floating point value parameter update is not notified from rqt_reconfigure but works from the command line in ROS2 foxy ?

jeremyfix, 2023-07-18:
Hello,

I'm using ROS2 foxy and facing the issue that a floating point value parameter changed in rqt_reconfigure does not get notified to the node which is declaring the parameter for specific values.

By default, I set the parameter to "0.5" . If I change it to "0.9", the change is not notified. If I change it to "1" , the change is notified.

Interestingly, the "." decimal point is not allowed (I cannot add it to the field although I strike the key) in the rqt reconfigure field, although it is the separator initially used to display the initial value. 

Changing the value of the parameter from the command line works, e.g. `ros2 param set param_issue linear_vel 1.5` from the code provided below.

Below is a minimal example for reproducing the issue :

```
import rclpy
from rclpy.node import Node
from rcl_interfaces.msg import SetParametersResult
from rcl_interfaces.msg import ParameterDescriptor
from example_interfaces.msg import Float32  # std_msgs.msg.Float32 is deprecated

class ParamIssue(Node):
    def __init__(self):
        super().__init__("param_issue")

        self.linear_vel = 0.5  # m/s
        self.declare_parameter(
            "linear_vel",
            self.linear_vel,
            ParameterDescriptor(description="The maximal linear velocity"),
        )
        self.add_on_set_parameters_callback(self.cb_params)

        self.vp_offset_sub = self.create_subscription(
            Float32, "vp_offset", self.on_vp_offset, 1
        )

    def cb_params(self, data):
        for p in data:
            name = p.name
            value = p.value
            setattr(self, name, value)
        self.get_logger().info(f"Parameter change notified : {self.linear_vel}")
        return SetParametersResult(successful=True)

    def on_vp_offset(self, msg):
        self.vp_offset = msg.data

def main(args=None):
    rclpy.init(args=args)

    paramissue = ParamIssue()
    rclpy.spin(paramissue)

    paramissue.destroy_node()
    rclpy.shutdown()
```

If you now start `rqt_reconfigure` , changing the value will either succeed or fail.

If you change it from the command line, this will always work .

Best;

Jeremy.

bmaxdk, 2023-08-12:
It seem odd. I think it could be the issue with `rqt_reconfigure` in your ros2 foxy..
Have you try reinstall `rqt_reconfigure`?
```bash
$ sudo apt update
$ sudo apt upgrade
$ sudo apt remove ros-foxy-rqt-reconfigure
$ sudo apt install ros-foxy-rqt-reconfigure
```

if this not works, could you try to declare your parameter as a string?
```python
self.declare_parameter("linear_vel_str", "0.5")

...

def cb_params(self, data):
    for p in data:
        name = p.name
        value = float(p.value) if name == "linear_vel_str" else p.value
        setattr(self, name, value)
    self.get_logger().info(f"Parameter change notified : {self.linear_vel}")
    return SetParametersResult(successful=True)
```

jeremyfix, 2024-03-12:
Unfortunately, I did not work on this question from last year. So I did not test the suggestion of @bmaxdk  on foxy.  And I cannot suggest you something else to test @phinners .