ros-visualization/rqt_reconfigure #142: Parameter list missing in reconfigure/dynamic if any parameter was declared without a default value under ROS2

EwingKang, 2024-02-15:
System: Ubuntu 22.04
Version: ROS 2 Iron Irwini - Patch Release 4

Following the parameter tutorial on the [ROS2 docs](https://docs.ros.org/en/iron/Tutorials/Intermediate/Monitoring-For-Parameter-Changes-CPP.html#prerequisites), I can see the following result in rqt/ rqt_reconfigure
[image]

However, if I add a parameter without default value using the [type-specifying version of the declare_parameter()](http://docs.ros.org/en/iron/p/rclcpp/generated/classrclcpp_1_1Node.html#_CPPv4N6rclcpp4Node17declare_parameterERKNSt6stringEN6rclcpp13ParameterTypeERKN14rcl_interfaces3msg19ParameterDescriptorEb) API, the entire parameter list for the node will disappear without any error message:
```c++
this->declare_parameter("an_int_param", 0); // The original demo
this->declare_parameter("an_int_param_without_val", rclcpp::PARAMETER_INTEGER);  // THIS CAUSE PROBLEMS
```
[image]

This problem disappears if that specific parameter is supplied with the command line:
```bash
ros2 run cpp_parameter_event_handler parameter_event_handler --ros-args -p an_int_param_without_val:=22
```
[image]

I should add that all other `ros2 param` and `ros2 service` command line tools are fine as far as I've tested.
This behavior is not mentioned anywhere in the documentation and I also couldn't find any similar problems on the internet. It took me a few hours of trial-and-error to figure-out the pattern.

EwingKang, 2024-08-23:
This is related to https://github.com/ros2/rclcpp/issues/2512

buckleytoby, 2025-01-03:
I'm also experiencing this issue, but to my knowledge my parameters ARE being declared with a default value: 

`n->declare_parameter<T>(name, def)`